Arian04/android-hid-client #13: IF YOU ARE USING THIS LET ME KNOW

Arian04, 2023-09-26:
I've stopped maintaining this because I just started working on other projects but I noticed a few stars so **if anyone is actually using this, let me know, I'll try to do some actual testing and iron out some kinks that I'm sure I left in the project before I forgot about it.**

404LiverNotFound, 2023-10-23:
I have been using it with KeePassDX as a password typer for a few months. Works great in my use case.

Arian04, 2023-10-23:
awesome, thanks for letting me know. now I have motivation to fix it up a bit lol

Arian04, 2023-10-28:
@404LiverNotFound any bugs you've noticed while using it? I found some old notes from when I was working on it and one of the issues I planned to address is that I don't make it clear to the user that enabling the character device (/dev/hidg0) necessary for the app to work disables special USB functions (USB tethering and all the other functions that show in that popup you get when you connect your phone to another device). I also don't have a builtin way to disable /dev/hidg0 inside the app to restore these functions. You could use USB gadget tool (another app) to do that but I'd rather it be builtin. Anything else you've run into?

Arian04, 2023-10-30:
interesting. it'd be cool if you could create an issue for it but it's not a big deal. I'll look into it soon, thanks!

Arian04, 2024-03-25:
closing because now I know I have a few users at least, so this is worth maintaining.

closed 2024-03-25 by Arian04

Arian04, 2024-03-26:
> I’m currently using the USB gadget tool to switch between character device and other USB functions. A built-in toggle would be nice since the only app I use that involves character device is this app. v1.0.0-beta.4 works well for me, but every key is sent with l_ctrl key pressed in v1.0.0. Should I open an issue regarding this?
> some logs from wev
> Keyboard: Hacker's keyboard (Gboard returns same result)

@404LiverNotFound v2.0.0 implements this feature. Pinging you because I thought you'd want to know.