Arian04/android-hid-client #14: L_CTRL sent with every key in v1.0.0 (regression from v1.0.0-beta.4)

Arian04, 2023-10-30:
**Describe the bug**
v1.0.0-beta.4 works well, but every key is sent with l_ctrl key pressed in v1.0.0.

**Device details (please complete the following information):**
 - Device: ?
 - OS: ?
 - Soft Keyboard: Hacker's Keyboard and Gboard exhibit the same behavior

**Logs (from <tt>wev</tt>)**
Input: <tt>A</tt>

Logs:
  ```
[14:     wl_keyboard] key: serial: 5965; time: 2827541; key: 37; state: 1 (pressed)
                      sym: Control_L    (65507), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 16
                      depressed: 00000004: Control 
                      latched: 00000000
                      locked: 00000010: Mod2 
[14:     wl_keyboard] key: serial: 5967; time: 2827541; key: 38; state: 1 (pressed)
                      sym: a            (97), utf8: ''
[14:     wl_keyboard] key: serial: 5968; time: 2827541; key: 37; state: 0 (released)
                      sym: Control_L    (65507), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 16
                      depressed: 00000000
                      latched: 00000000
                      locked: 00000010: Mod2 
[14:     wl_keyboard] key: serial: 5970; time: 2827541; key: 38; state: 0 (released)
                      sym: a            (97), utf8: ''
[14:     wl_keyboard] key: serial: 5971; time: 2827542; key: 37; state: 1 (pressed)
                      sym: Control_L    (65507), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 16
                      depressed: 00000004: Control 
                      latched: 00000000
                      locked: 00000010: Mod2 
[14:     wl_keyboard] key: serial: 5973; time: 2827544; key: 37; state: 0 (released)
                      sym: Control_L    (65507), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 16
                      depressed: 00000000
                      latched: 00000000
                      locked: 00000010: Mod2

  ```

Arian04, 2023-10-30:
I can't reproduce this.

**Details**
Phone: Pixel 5, running lineageOS 20
Computer: Archlinux, up to date as of yesterday. I believe key events are handling by the compositor on Wayland, so I'll include that I'm using Kwin (KDE Plasma).
Keyboard: Hacker's Keyboard
App Version: I grabbed the v1.0.0 APK straight off the releases page and fully uninstalled my previous version of the app and installed the fresh APK 

<details>
    <summary>My <tt>wev</tt> logs (Annotated, nothing ommitted)</summary>

```
[07:         wl_seat] capabilities: pointer keyboard touch
[07:         wl_seat] name:
[10:    xdg_toplevel] configure: width: 0; height: 0
[09:     xdg_surface] configure: serial: 93390
[14:     wl_keyboard] repeat_info: rate: 25 keys/sec; delay: 660 ms
[14:     wl_keyboard] keymap: format: 1 (xkb v1), size: 64756
[13:      wl_pointer] enter: serial: 93391; surface: 3, x, y: 596.000000, 479.000000
[13:      wl_pointer] frame
[14:     wl_keyboard] enter: serial: 93395; surface: 3
'                     sym: Return       (65293), utf8: '
[14:     wl_keyboard] modifiers: serial: 0; group: 0
depressed: 00000000
latched: 00000000
locked: 00000000
[11:  wl_data_device] data_offer: id: 4278190080
[4278190080:   wl_data_offer] offer: mime_type: text/plain
[4278190080:   wl_data_offer] offer: mime_type: application/x-kde-onlyReplaceEmpty
[4278190080:   wl_data_offer] offer: mime_type: text/plain;charset=utf-8
[4278190080:   wl_data_offer] source_actions: actions: 0 (none)
[11:  wl_data_device] selection: id: 4278190080
[10:    xdg_toplevel] configure: width: 640; height: 480
activated
[09:     xdg_surface] configure: serial: 93398
[14:     wl_keyboard] key: serial: 93399; time: 151600607; key: 36; state: 0 (released)
sym: Return       (65293), utf8: ''

```
Everything above here is just from it starting up. The Return release was ofc from me releasing Enter after running the command

```
[14:     wl_keyboard] key: serial: 93400; time: 151610902; key: 50; state: 1 (pressed)
sym: Shift_L      (65505), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 0
depressed: 00000001: Shift
latched: 00000000
locked: 00000000
[14:     wl_keyboard] key: serial: 93402; time: 151610902; key: 38; state: 1 (pressed)
sym: A            (65), utf8: 'A'
[14:     wl_keyboard] key: serial: 93403; time: 151610904; key: 50; state: 0 (released)
sym: Shift_L      (65505), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 0
depressed: 00000000
latched: 00000000
locked: 00000000
[14:     wl_keyboard] key: serial: 93405; time: 151610904; key: 38; state: 0 (released)
sym: a            (97), utf8: ''

```
The above output was from sending a capital A (only `A`) using Direct Input (I hit Shift, then A on Hacker's Keyboard).

Below, I typed an `A` into Manual Input and sent it. This should be soft keyboard independent.

```
[14:     wl_keyboard] key: serial: 93406; time: 151614420; key: 50; state: 1 (pressed)
sym: Shift_L      (65505), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 0
depressed: 00000001: Shift
latched: 00000000
locked: 00000000
[14:     wl_keyboard] key: serial: 93408; time: 151614420; key: 38; state: 1 (pressed)
sym: A            (65), utf8: 'A'
[14:     wl_keyboard] key: serial: 93409; time: 151614422; key: 50; state: 0 (released)
sym: Shift_L      (65505), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 0
depressed: 00000000
latched: 00000000
locked: 00000000
[14:     wl_keyboard] key: serial: 93411; time: 151614422; key: 38; state: 0 (released)
sym: a            (97), utf8: ''

```

Output seems the same between Direct and Manual input modes.

</details>

As you can see, I can't reproduce the issue. I tried to very thorough in explaining my exact process, so if you notice that way you reproduced the issue was different from the way I attempted to, let me know.

Arian04, 2023-10-30:
ideas: 
- maybe USB gadget tool is messing with the report descriptor, if it's not too much trouble, maybe try rebooting (to clear out the old character device) and open my app, then create the device and see if the issue still occurs.
- Could I have some details on the target computer? (Distro, keyboard layout, etc.) just in case it's necessary
the (Arch Wiki)[https://wiki.archlinux.org/title/Linux_console/Keyboard_configuration#Viewing_keyboard_settings] recommends the following to see your current keyboard settings:

```
$ localectl status
System Locale: LANG=en_US.UTF-8
VC Keymap: (unset)
X11 Layout: (unset)
```
As you can see I also included the output I got so we can compare.

Arian04, 2023-10-31:
Just had another idea, did you change the CTRL setting that I mention in the readme?

404LiverNotFound, 2023-11-01:
I did change the settings in README. After further testing, I believe the issue here is creating the character device with USB gadget tool in the first place. Everything works as intended only when I create a character device in the app. Switching between the default gadget and the keyboard gadget in USB gadget tool afterwards does not cause any problems.

**Details**
- App Version: v1.0.0 from Github release page
- Phone: Sony Xperia 1 II, latest stock firmware (Android 12)
- OS: Archlinux + KDE Wayland session, with full system upgraded in 2023/10/31
  - Also tested on bare metal Windows 10. Both show same behavior.
- Keyboard: Hacker's Keyboard, with README CTRL settings
- Locale: 
  ```
  $ localectl status
  System Locale: LANG=en_US.UTF-8
  VC Keymap: (unset)
  X11 Layout: us
  ```

Arian04, 2023-11-01:
yup I create the char device in a special way that's different than that app does it. I did it that way so mine can send extra keys like multimedia keys. glad we could figure it out