Arian04/android-hid-client #16: versioning issue

IzzySoft, 2024-03-25:
The APK at the latest release (v1.1.0) identifies as

```
package: name='me.arianb.usb_hid_client' versionCode='1' versionName='1.0'
```

Looks like you forgot to [update that in `build.gradle`](https://github.com/Arian04/android-hid-client/blob/main/app/build.gradle#L12). Hard to tell the versions apart then :wink: Can you please keep them "in sync"? `versionCode` should simply be increased with each new release (Android uses that internally to tell versions apart and to decide whether an APK is an update for the installed app), `versionName` should ideally match the tag/release name.

Thanks in advance!

Arian04, 2024-03-25:
you're so right lol I completely forgot, I'll fix that and update the APK asap. when I made all those changes I wasn't making frequent atomic commits (shame on me) so I had to de-tangle my massively changed working tree and turn it into a bunch of smaller commits. must've missed that change, I'll fix it ASAP, thank you!

Arian04, 2024-03-25:
just updated the release asset, thanks again for letting me know!

still figuring out how I wanna handle committing it since the main branch is a WIP v2.0.0. I really should adopt a proper git workflow and set up release branches or something. Maybe now is the time.

also I just realized you're Izzy as in IzzyOnDroid! I Love that repo and ty for maintaining it!

IzzySoft, 2024-03-25:
> I just realized you're Izzy as in IzzyOnDroid! I Love that repo and ty for maintaining it!

Thanks! :star_struck: Guess you have a clue where my question and your swift action leads to now then, right? :yum:

Arian04, 2024-03-25:
omg were you planning on adding it to the repo? that's awesome!!

IzzySoft, 2024-03-25:
Sure! Almost done. One little thing:

    ! repo/me.arianb.usb_hid_client_2.apk contains signature block blobs: 0x504b4453 (DEPENDENCY_INFO_BLOCK; GOOGLE)

Can be avoided easily:

```gradle
android {
    dependenciesInfo {
        // Disables dependency metadata when building APKs.
        includeInApk = false
        // Disables dependency metadata when building Android App Bundles.
        includeInBundle = false
    }
}
```

For some background: that BLOB is supposed to be just a binary representation of your app's dependency tree. But as it's encrypted with a public key belonging to Google, only Google can read it – and nobody else can even verify what it really contains.

IzzySoft, 2024-03-25:
OK, this goes live at around 7 pm UTC (i.e. in less than 2h):

[image]

`DEPENDENCY_INFO_BLOCK` can be solved with the next release :wink: Welcome aboard! :smiley:

Arian04, 2024-03-25:
awesome, ty for the information and ty again for adding me!! Just a note, I've been messing with the branches and a bit of the history to make it so `main` is prod-ready code and not just whatever I'm working on, so the README has just updated (moved back in time by a few commmits). Just letting you know since I wouldn't want any users to be misled in case they see that description, install the app, and get confused about why there actually *isn't* yet mouse support (in v1.1.0).

IzzySoft, 2024-04-05:
Oof, must have been one of the "lost notification mails" – so apologies for the late reply…

> in case they see that description, install the app, and get confused about why there actually isn't yet mouse support

There's a solution for many things. For things like that, may I recommend you establishing Fastlane structures here? I could send you a PR with what's currently set up in my repo, and you could use my [Fastlane Cheat Sheet](https://gitlab.com/-/snippets/1895688) to build on that. Which would give you control about how your app is presented, making sure the information matches the app and the screenshots etc. do as well.

Arian04, 2024-04-05:
it's all good, since I last sent the message, I've merged the mouse support to main. I've also started using a "develop" branch for unreleased code and merge to "main" right before every release, so hopefully it wouldn't be an issue anymore.

but I'll definitely check out that link when I get a chance, thanks!

IzzySoft, 2024-04-05:
My offer stands. Once you've checked the link and think it would work for you, I can send you a "starter package" – just let me know then :smiley:

Arian04, 2024-04-05:
I just got a chance to look it over really quick, and it looks great! if you get a chance I'd really appreciate the PR. but if your fastlane config is in a public repo, then I could just grab that myself if you give me a link (couldn't find it when I looked earlier).

I've been busy lately though, so it might take a bit before I can get to merging and updating it.

IzzySoft, 2024-04-06:
> I'd really appreciate the PR

then let me send it. No big deal, takes me maybe 5 minutes.

> but if your fastlane config is in a public repo

Just not in a public *git* repo, so it would be harder for you to pick things and take 5..10 times longer than getting the PR ready.

> I've been busy lately though, so it might take a bit before I can get to merging and updating it.

No prob. Wouldn't be pulled before the next release anyway, unless I'd trigger it manually.

Arian04, 2024-04-06:
perfect, thank you!!!