Arian04/android-hid-client #25: App doesn't work

srt54558, 2024-03-29:
**Describe the bug**
Hello, the app always says "device not connected" (even tho it is) and when exporting debug log it says "error getting logs". The app didn't manage to make the phone a HID device so I used "USB gadget tool" (which I like to use more). No keys are being sent to the pc

**To Reproduce**
Steps to reproduce the behavior:
1. Open app
2. Try anything 
4. See error

**Expected behavior**
The app should work as intended; be a keyboard app

**Screenshots**
It's only showing the built in error messages

**Device details (please complete the following information):**
 - Device: Google Pixel 8 Pro 
 - OS: Android 14 QPR2 (March update)
 - Soft Keyboard: Gboard
 - Are You Rooted? Yes
 - Root Method: Magisk 27.0

**Additional context**
Keyboard emulation works on "Rucky" (a app for running ducky scripts)

I love the idea and I'm so happy someone did it because the old app "USB keyboard" doesn't work (Pixel 8 Pro has no 32bit support)

Arian04, 2024-03-30:
working on some improvements right now that'll make it easier to debug some issues like this. After I get those finished I'll follow up on this.

Windows81, 2024-04-07:
Try this [debug build](https://github.com/Arian04/android-hid-client/actions/runs/8587737192/artifacts/1391780146) (link probably won't last forever) out.  I encountered a very similar problem, so you might be in luck.

Arian04, 2024-04-07:
i was actually wondering if this was the same issue, thanks for linking the build! If it works then we have our answer, but if not then I'm working on improved debugging right now.

Arian04, 2024-04-13:
haven't made as many improvements to debugging as I planned to since I've been pretty busy but can you uninstall the app (if you still have it installed), install the latest one from the Github releases page, try it, and if it's still broken, export logs from the bottom of the settings page and attach them here.

Windows81, 2024-04-14:
Doesn't work.  I don't see any HID fallback option in the settings.  Before I first boot the app, **I already have had added `hid.keyboard` and `hid.mouse` to `g2` through USB Gadget Tool**.  After I launch the app and fix character permissions, there is no observed change.  Additionally, all the other functions of `g1` and `g2` work normally.

[debug_log_me.arianb.usb_hid_client_1713067249.txt](https://github.com/Arian04/android-hid-client/files/14969444/debug_log_me.arianb.usb_hid_client_1713067249.txt)

Arian04, 2024-04-14:
yea i havent added that option yet, my reply was meant for the author of this issue

Arian04, 2024-06-30:
I've improved the ability to troubleshoot issues like this in the latest release (v2.3.0). Do you mind trying that out and exporting a log file for me to check out?

You would just have to:
1. Install v2.3.0
2. Give the app root permissions
3. Go to the "Troubleshooting" screen in the overflow menu in the top right
4. Click the button at the bottom to export logs (a screenshot of the troubleshooting page would also be nice)

luimu64, 2024-07-09:
Same issue, just complains about not being connected. Used 2.3.0. Below logs and screenshot of the troubleshooting page.

[debug_log_me.arianb.usb_hid_client_1720556658.txt](https://github.com/user-attachments/files/16149921/debug_log_me.arianb.usb_hid_client_1720556658.txt)
[image]

Arian04, 2024-07-09:
thanks for the logs! unfortunately I was a little bit conservative on how large I allowed the log buffer to be, so it dropped the earlier logs which contained the actual character device creation script output. I'll fix that soon, but from what I *could* see, everything looks like it should work. the "device unplugged" error says that, because in my experience, the underlying error is generally caused by the device being unplugged, but the error itself could indicate other problems.

so basically (as I've previously seen mentioned in [tejado/android-usb-gadget](https://github.com/tejado/android-usb-gadget) issues), this seems like a very weird issue thats gonna take some more troubleshooting. I'll do some thinking and research soon probably

Arian04, 2025-06-06:
Hey everyone, I just released v3.0.0-beta, so I thought I should ping people. As explained in the release notes, I added support for some "device-specific quirks" and I also slightly modified the gadget configuration process.

Basically what I'm saying is:
1. please try out the v3.0.0-beta (following the update procedure at the top of the release notes)
2. see if there's still a problem
3. if so, then try to enable "Experimental mode", then enable the "Disable gadget functions during configuration" option that shows up underneath it, then reopen the app and try again.

At that point, if your issue is gone, please let me know, and if your issue is NOT gone, also please let me know.

These commits have fixed a few people's issues already, so I'm wondering how many other people's issues will be resolved by these changes.

Thank you!

CiLz91, 2025-07-02:
Not working, double check all settings including root permission, even using usb gadget apps, not working at all, connected but not as hid..
Xperia 1 j9110
Android 11
Latest Magisk

Arian04, 2025-07-07:
> > [@CiLz91](https://github.com/CiLz91): rebooting is supposed to reset all your HID settings.
> > 
> > Did you test previous versions of the app as well?
> 
> Ofc sir, not working at all
> Tested with Xperia 1 rooted and Mipad 2 Tpm F12 and my Nokia 808 as well hehe

wow 3 different devices are all not working with it? either there's an issue on your side or that's some crazy bad luck. bad luck is definitely a possibility though because those sound like the types of devices that exhibit issues w HID gadget functions

Arian04, 2026-09-19:
Hi everyone, i'm working on some old issues again and I think this issue is getting a bit difficult to track, since it contains multiple unrelated issues inside of it. Here's my summary of the different issues so far:

user srt54558 (status: ?)
- issue author, no replies after issue creation + a reaction (app version: unsure, but before v2.3.0)
- issue: "not connected"
- device info included, but no logs (i don't think the log feature existed yet)

user Windows81 (status: hopefully solved)
- made comment under issue
- issue: gadget path stuff
- I think this was the issue tracked in #38, which has been solved, so hopefully things should work for this person now

user luimu64 (status: ?)
- made comment under issue, no replies after initial comment (app version: v2.3.0)
- issue: "not connected", same as issue author, troubleshooting info all seems like things should work properly
- logs attached

user CiLz91 (status: ?)
- made comment under issue
- issue: unknown "Not working... not working at all, troubleshoot and disable trouble shoot, including reboot, connected but not as hid.."
- no logs

user codesktop (status: solved)
- made comment under issue
- issue: touchpad movement issue, not device-specific

---

I think I'll close this one and split out the issues into more specific versions, so that it's easier for me to keep track of (and contact people for more info if desired).

My plan right now is to split it into one issue for "device possibly unplugged / not connected" (for srt54558 and luimu64), then I will ping CiLz91 and Windows81, asking if they still care for this issue, and if so, then I will create new issues for them as well and look into those.

Actually as I'm writing this, I'm thinking I'll just ping them here.

Hi @CiLz91 and @Windows81, please read the message above and let me know if either of you are interested in looking into this issue again. if so, I will create new issue for you (if things are still not working). No worries if you are no longer interested in this.

Arian04, 2026-09-30:
Actually instead of splitting this into issues that may no longer be relevant, I'll just close this, and if anyone cares about reopening this, let me know and we can create a fresh issue.

closed 2026-09-30 by Arian04