Arian04/android-hid-client #27: App doesn't open in v2.0.0 and V2.1.0

Gabrick75, 2024-04-03:
**Describe the bug**
I try to open the application, but it doesn't open :/

**To Reproduce**
Steps to reproduce the behavior:
1. Install the app
2. Click on the app icon
3. The app doesn't open
4. If you try too many times, the standard Android error message appears.

**Expected behavior**
The application is expected to open normally 

**Screenshots**
I believe there is no need to add screenshots, But if you need it I'll add it later

**Device details:**
 - Device: Motorola moto G9 Play
 - OS: Stock Android 11
 - Soft Keyboard: Hacker's Keyboard and google keyboard 
 - Are You Rooted? Yes
 - Root Method: Magisk 26.3

**Additional context**
Version V1.1.0 works normally on my device.

So far the app has been wonderful for me, it saved me a lot of trouble with my PC! I can't wait to get started on these new updates with the addition of the Mouse function

If you need help translating, I can help by translating into Brazilian Portuguese(pt-br)

Thanks for the app so far

Arian04, 2024-04-06:
thanks for the bug report, just wanted to let you know I reproduced this issue. I'm a pretty busy right now, but I'm working on setting up automated testing of my app in emulators so that version-specific issues like this are much less likely to happen (since I currently only test on my main phone).

Gabrick75, 2024-04-06:
Right! Any information you need and I can help, just let me know.

Arian04, 2024-04-07:
I pushed the fix (c2e2866) a few hours ago but got caught up dealing with a different issue and forgot to send this comment until now.

It might take me a little while to get the fix into the next release, but as long as you're okay with dealing with a debug APK, then [this apk](https://github.com/Arian04/android-hid-client/actions/runs/8585258139/artifacts/1391299076) should fix it. If not then this will definitely be fixed in the next release like I said.
 
It's a debug APK so you'll have to uninstall the version of the app you already have installed. You'll also have to uninstall and re-install later on when this gets officially released.

closed 2024-04-07 by Arian04

Arian04, 2024-04-07:
oh also, if that doesn't fix it for you, we can reopen this issue