Arian04/android-hid-client #29: Versions after 1.0.0+ don't work on Surface Duo [Android 12]; pull request might be proposed soon.

Windows81, 2024-04-07:
**Describe the bug**
I use a Surface Duo 1.  The last working version for me is 1.0.0.

**To Reproduce**
Steps to reproduce the behavior:
1. Before I launch the app, I use [USB Gadget Tool](https://github.com/tejado/android-usb-gadget) and create `hid.keyboard` and `hid.mouse` functions on Gadget path `/config/usb_gadget/g2/`. 
 `/config/usb_gadget/g2/` points to a Gadget with no functions pre-defined and is second on the list when you launch USB Gadget Tool.

1. I open your USB HID Client and continue past initial setup.

1. I receive a popup saying that `/dev/hidg0` doesn't exist and accept the program's suggested fixes.

1. I refresh USB Gadget Tools and discover that there are new functions in the wrong Gadget `.../g1/`.  I still need to clarify this point.

1. I receiving a popup which says that my character permissions might not be set up properly.

1. I tap 'FIX'; trace to previous step.

**Device details (please complete the following information):**
 - Device: Surface Duo 1
 - OS: Android 12
 - Soft Keyboard: SwiftKey
 - Are You Rooted?: Yes/No
 - Root Method: Magisk

**Additional context**
Add any other context about the problem here.

Arian04, 2024-04-07:
when I first wrote the code, I tried looking for cases of devices with gadget names besides `g1` and couldn't find any, so I hardcoded it. I guess now it's time to write some logic for that.

would appreciate the PR but I could also handle it pretty easily, just been busy lately with stuff in real life and a decent amount of things changes I wanna complete for this project, so it might take a bit until I can make a release.

Windows81, 2024-04-07:
No problem!

If I were you, I would have the user select which gadget to use upon initial setup.

I'm not yet familiar with the underlying routines behind USB Gadget Tool, but I'd reckon it's easy to look into.

On my device, `/dev/hidg0` and `/dev/hidg1` are only usable when I use them on `g2`.

Arian04, 2024-04-07:
I'd rather have it be entirely transparent to the user, but yea if the detection logic is too complex I'll just present a dropdown of all discovered gadgets or something.

Arian04, 2024-04-07:
It might take me a little while to get the fix into the next release, but as long as you're okay with dealing with a debug APK, then [this apk](https://github.com/Arian04/android-hid-client/actions/runs/8587737192/artifacts/1391780146) should fix it.
 
Since it's a debug APK (meaning it's not signed with my main signing key) you'll have to uninstall the version of the app you already have installed. You'll also have to uninstall and re-install later on when this gets officially released.

If using a debug APK until I can get the next release ready bothers you, I can build a proper signed version and upload it here.

closing this issue, but if the fix doesn't work for you, then we can definitely reopen this.

Windows81, 2024-04-07:
Will do.  I appreciate the help!

Windows81, 2024-04-07:
Touchpad: works.
Manual input: works.
Direcyt input: works.

Windows81, 2024-04-07:
https://github.com/Arian04/android-hid-client/files/14896693/apk.zip

I'll leave the `.apk` file as an attachment for posterity.

Arian04, 2024-04-07:
backspace and modifier keys not working is strange. Did you add HID functions using USB gadget tool? because my app uses a different report descriptor to enable extra features like multimedia keys, so weird things like that will happen if the functions are added using a different app.

Windows81, 2024-04-08:
My bad.  I use Swiftkey, so the backspace input might be different.  I didn't find the modifier keys myself, but there's a chace that I didn't look well enough.

Arian04, 2024-04-08:
not sure if you meant you couldn't find it in the app or in Swiftkey, but the app doesn't provide modifier key buttons, it just uses your normal keyboard. that's why I recommend Hackers Keyboard to give users access to way more keys.

and I'll see if I can start testing out a variety of keyboards automatically in my instrumented tests. Thanks!

Windows81, 2024-04-08:
Actually, for each character I type with direct input, I need to type that many backspaces before it begins to work.

For example, when I type 'AB' in direct input, I tap backspace three times and then the string becomes 'A'.  When I type 'abcd', I must hit the backspace key five times before it becomes 'abc', then one time after that before it becomes 'ab'.

This buffering behaviour doesn't apply when the characters I've typed are *not* from direct input.

Arian04, 2024-04-08:
that's incredibly strange. I should definitely test on a wider variety of keyboards. I just haven't because I only recently began setting up automated testing.

thanks for the info!!

Windows81, 2024-04-08:
No problem.  I really think that `hid.keyboard` and `hid.mouse` should be added to `.../g2/`.  Otherwise, other USB functions on `.../g1/` won't work at all.

Having a toggle would help as well.

Arian04, 2024-04-08:
wait are you saying you have a g1 AND g2?

Windows81, 2024-04-08:
[image]

Yes.  I have a `g1` **and** a `g2`.

Oddly enough, every app except for this one works with the configuration I've got set up above.

Arian04, 2024-04-08:
this is incredibly weird. so while g2 is shown as "active" you can use all your standard Android gadget functions (listed under g1) but if HID functions are added to g1 and g1 is activated, you can no longer use the standard Android gadget functions? unless I'm misunderstanding this sounds backwards.

Windows81, 2024-04-08:
Let me fix the table.

Windows81, 2024-04-08:
Done.

Windows81, 2024-04-08:
Not quite.  `g2` has the keyboard stuff.  ADB won't work when I have `g2` active.

Arian04, 2024-04-08:
that's so strange. the reason I changed the app to modify the default Android gadget rather than creating a custom one is because on my device, *all* features work simultaneously that way.

now that I know what your situation is, I see why you wanted me to offer a choice of which gadget to modify. I'll put that on the list.

also, regarding your removal note, there is a way to remove the gadgets or functions but USB gadget tool doesn't expose it. it also causes a kernel panic on my device ever since updating to android 14, so heads up on that. if you don't mind going in a terminal youd just have to use a script like [this](https://github.com/larsks/systemd-usb-gadget/blob/master/remove-gadget.sh) but with the path at the top changed to the correct one.

Windows81, 2024-04-08:
It's a good thing that Android kept that bit of the kernel intact.

Windows81, 2024-04-08:
Weird.  I tried the whole process again and saw dfferent stuff.  I previously stated that ADB shell is also affected, which I didn't find this time around.

When `hid.*` is supplied to `g1`, ADB works, but USB file transfer and tethering are broken until I reboot.

I'll revise the previous posts.

Arian04, 2024-04-08:
interesting! this time around, did you let my app add the HID functions or did you do it through USB gadget tool?

Windows81, 2024-04-08:
I used your app.

Windows81, 2024-04-08:
I did this:
1. Booted.
2. Connected to USB.
3. Enabled system-stock USB file transfer.
4. Opened USB Gadget Tool and added `hid.*` to `g2`.
5. Activated `g2`.
6. **Successfully** wsed another keyboard app.
7. Re-activated `g1`.
8. **Successfully** opened system-stock file-transfer again.

Arian04, 2024-04-08:
yea I'll just eventually add an option to add the functions to a specific gadget if there are multiple

Arian04, 2024-04-09:
accidentally closed while i did some extremely messy merging because that old message was from when I didn't fully understand this issue

Arian04, 2024-04-24:
closing this issue in favor of #38, so the title can be more descriptive and I don't forget to add this.

commit 8a36d3ee (2024-04-07): fix: handle gadgets that aren't named g1

fixes #29
commit 338bb2e4 (2024-04-09): fix: handle gadgets that aren't named g1

fixes #29
closed 2024-04-24 by Arian04