Arian04/android-hid-client #31: Add two finger scroll functionality to touchpad

Arian04, 2024-04-13:


Arian04, 2024-06-19:
I had a thought a while ago that I should document here.

Don't operating systems usually handle gestures on their own? So maybe I just need to send proper events for finger up/down and ensure that it works for multiple fingers at the same time, then the OS can figure it out? I think? If that's all true, I'll probably just make it compliant with a standard precision touchpad spec and it should just work™

brian6932, 2024-06-20:
I believe on Windows you'd have to have a Windows Precision Driver to use the native touchpad features https://learn.microsoft.com/windows-hardware/design/component-guidelines/touchpad-implementation-guide.

Arian04, 2024-07-02:
implemented in 66db628dadfcd1c65e3261863319fee75caa1a3e and will be available in the next release.

closed 2024-07-02 by Arian04