Arian04/android-hid-client #33: App doesn't work (maybe because I'm Android 13 but come on I really need it)

Key1569, 2024-04-17:
I'm rooted with magisk and when I'm going to create character device functions it doesn't work

"ERROR: Failed to create character device"

Arian04, 2024-04-17:
device model? custom rom or stock android?

Key1569, 2024-04-18:
> device model? custom rom or stock android?

Galaxy A03 One UI 5.1

n00b69, 2024-04-23:
Same issue on Galaxy S10 OneUI 5.1

tamilan007, 2024-06-19:
Same Issue Galaxy A03 core 
android 11 go with magisk root
stock rom

quintus3236, 2024-06-22:
Same issue here.
Redmi Note 9
LineageOS 20, Android 13

Arian04, 2024-06-30:
I've improved the ability to troubleshoot issues like this in the latest release (v2.3.0). Do you all mind trying that out and exporting a log file for me to check out?

You would just have to:
1. Install v2.3.0
2. Give the app root permissions
3. Go to the "Troubleshooting" screen in the overflow menu in the top right
4. Click the button at the bottom to export logs (a screenshot of the troubleshooting page would also be nice)

Arian04, 2024-08-03:
closing due to lack of response. will reopen if there's a response later on.

Key1569, 2024-08-03:
> closing due to lack of response. will reopen if there's a response later on.

[debug_log_me.arianb.usb_hid_client_1722716235.txt](https://github.com/user-attachments/files/16482781/debug_log_me.arianb.usb_hid_client_1722716235.txt)

Arian04, 2024-08-03:
your kernel doesn't have configfs HID function support. you gotta build your kernel with support enabled since it isn't already enabled.

closing this issue as solved. if anyone other than the author of this issue wants to respond with their logs, please make your own issue for organization purposes.

closed 2024-08-03 by Arian04