Arian04/android-hid-client #40: Send button disappears

gaussandhisgun, 2024-04-27:
**Describe the bug**
The send button next to the multiline input disappears unless the field itself is two lines high or more.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the app
2. See the issue

**Expected behavior**
The send button is there anytime I might need it.

**Screenshots**
[image]

**Device details (please complete the following information):**
 - Device: Redmi Note 9S
 - OS: Android 14 (GhostOS)
 - Soft Keyboard: GBoard
 - Are You Rooted? Yes
 - Root Method: Magisk 

**Additional context**
`wm size`: 1080x2400
`wm density`: 378

Arian04, 2024-04-27:
thanks so much for all the details!! I've been really busy lately but just saw how many details you included and wanted to say that I'll try to reproduce this as soon as i have more time.

Arian04, 2024-06-30:
I was unable to reproduce this on v2.2.1, but the UI part of the app has been pretty much entirely re-written since then. Could you test the latest release (v2.3.0) and let me know if this still happens? Thanks!

gaussandhisgun, 2024-07-06:
yeah, it seems to be perfectly fine in its current state

Arian04, 2024-07-06:
great, thanks!

closed 2024-07-06 by Arian04