Arian04/android-hid-client #42: Add a loopback mode to use phone screen as trackpad for attached second screen

daniellandau, 2024-05-12:
It would be great if the code in this app could be easily extended to not send mouse movements out from the USB, but instead register to the local Android system as a connected emulated external mouse and that way the primary phone display could be used as a trackpad for an attached external display. When there's an external display, a USB mouse automatically shows a cursor only on the external display (when using Force Desktop mode from developer settings), but for example KDE Connect mouse receiver only shows a cursor on the primary phone display.

Arian04, 2024-05-12:
interesting use case! so just to confirm, when you connect a USB mouse to the phone, you're able to simultaneously use the mouse on the external display and your fingers on the touchscreen of the phone itself?

daniellandau, 2024-05-13:
Yes, although it seems to be dependent on the order of plugging the display and the mouse in. Sometimes it goes to a state where the mouse cursor is also stuck on the phone screen. This is on Android 14 crDroid.

Arian04, 2024-05-16:
interesting!! thanks for the info, I'll look into this soon. slowly getting back into working on this after being busy

Arian04, 2024-05-20:
Could you attempt to use adb [as described here](https://stackoverflow.com/questions/61346487/simulate-usb-mouse-pointer-input-on-android-phone) to send inputs like you want to?

If it's not too much trouble, could you also check if `/dev/uhid` is present on your system?

Just looking into which implementations of this are possible/most convenient right now. Thanks!

gaussandhisgun, 2024-06-04:
fyi: this can be tested without a physical external display if you enable both Force Desktop Mode (android 10+) and Emulate External Displays settings in Developer Settings on your phone

Arian04, 2024-06-04:
thanks! I actually looked for that earlier but ig somehow missed it, because I found it right now. might've not noticed a typo when I searched + just not seen it when I scrolled through settings.

Arian04, 2024-07-02:
just wanted to update on this, I've confirmed that communicating with /dev/uhid allows me to control a cursor on a simulated external display. I only tested it with a C script to see if it was possible, so I still gotta implement this in the actual app, but I just wanted to let you know that it does seem possible.

@daniellandau again, please let me know if /dev/uhid is present on your device, otherwise it would be pretty sad if I implemented this feature only for you to not be able to use it lol.

daniellandau, 2024-07-02:
Ah sorry, I've been absent. I do have /dev/uhid. Is the C source code somewhere so I could check with a physical display?

Arian04, 2024-07-02:
great! I just used the sample UHID code from the linux kernel source code [here](https://github.com/torvalds/linux/blob/master/samples/uhid/uhid-example.c). I compiled it on my device in termux after installing the `clang` package. then I ran it from within termux using `sudo` from the `tsu` package, but I'm sure you could just `su` then run it normally if you don't have that package. The comments at the top of the source file explain how the script can be used, but the gist is just use WASD and see if your cursor moves around.

daniellandau, 2024-07-03:
Yes, that works. Same as with physical mouse, the cursor sometimes gets stuck on phone screen, but most probably that needs to be fixed on Android side.

Arian04, 2024-08-09:
got busy so it took me a while to get to this, but it's implemented in 2e71bcd4ebd198f17ea8861264aee5587be766f8. It was a bit more involved than I expected, but it works. Heads up, for a reason I've yet to discover, once enabled, it will take several presses/swipes/interactions for it to actually start working, but it seems to work flawlessly after that. I've been testing on a simulated external display, so if it ends up being broken on your device, let me know.

CI built the commit here: https://github.com/Arian04/android-hid-client/actions/runs/10311477553/artifacts/1793032171

it'll download as a zip file, so just extract it before installing. It's not signed with my main key, so you'll have to uninstall the previous version before installing this version.

closed 2024-08-09 by Arian04