Arian04/android-hid-client #43: Feature Request: Discrete left and right click buttons

ThatStella7922, 2024-06-04:
**Is your feature request related to a problem? Please describe.**
Currently, I don't seem to have a way to right click and I'd like to be able to click and drag things.
Discrete left and right click buttons would fix these issues.

[image]
Something like this. (Separate scrollable area would be nice in addition to the two finger scroll that's planned(

Arian04, 2024-06-05:
planning to add settings for standard touchpad "gestures" (3 finger tap middle click, 2 finger tap right click, 1 finger tap left click, two finger scroll, etc.) or discrete buttons like you showed. Thanks for the nice mockup!

gaussandhisgun, 2024-06-08:
that's not a mockup, that's https://play.google.com/store/apps/details?id=io.appground.blek - basically the same concept, except using Bluetooth and full of microtransactions

Arian04, 2026-08-16:
Closing since I added this for the default Touchpad implementation in v3.1.0

closed 2026-08-16 by Arian04