Arian04/android-hid-client #47: Only touchpad option

vacuumbeef, 2024-07-09:
It would be really cool to have an option to have just a touchpad on the screen - you could put your phone under the keyboard, for example, and have a basically laptop experience. Right now, in horizontal position touchpad area only takes the bottom half of the screen, so it's not really usable.

<details>
  <summary>Screenshot</summary>
  
[image]
  
</details>

Well it's my personal immediate wish, I don't know if anyone else would find it a good idea.

Arian04, 2024-07-09:
good idea, I was actually getting annoyed about this earlier while trying to use it horizontally. I just need to figure out the least clunky way to implement this, because I'm not sure if this should be a setting, or if it should just be simple to "expand"/hide the manual input text box + send button

vacuumbeef, 2024-07-09:
Personally I see the expand/hide button or something as the best way

Arian04, 2024-08-13:
Implemented in the commit linked above. I also fixed up some general spacing issues in addition to adding this feature. Here's a screenshot with this feature enabled for reference:

[image]

closed 2024-08-13 by Arian04

vacuumbeef, 2024-08-13:
I'd say it's even better.

commit bd560f4d (2024-08-13): implement setting that hides Manual Input when in landscape to make more space for the Touchpad

Closes #47