Arian04/android-hid-client #50: Failed to read permissions

Damglador, 2024-09-25:
**Describe the bug**
App doesn't work, I mean it doesn't pass keys and trackpad. Every time I return to main screen of the app the popup pops up with No or Yes (I press Yes every time). Maybe it faild to add HID thing.

In troubleshoot screen:
- Chracter Device Info (Both) errors
- Error log: `Failed to read permissions `

Root permissions are available through KernelSU, the command `ksud sepolicy patch` works in Termux (I didn't pass any parameters, but it gives an output)

**To Reproduce**
Steps to reproduce the behavior:
1. Install app
2. Grant roo
3. Open the app
4. Click yes
5. Observe how nothing works

**Expected behavior**
I guess everything in troubleshooting should be checked and app should be working

**Screenshots**
[image]

**Device details (please complete the following information):**
 - Device: Ulefone Armor 21
 - OS: Android 14
 - Soft Keyboard: GBoard
 - Are You Rooted? Yup
 - Root Method: KernelSU 

**Additional context**
Da loog
[image]

Maybe I should do anything to make it work?

Arian04, 2024-09-25:
well that's a strange issue. every time I think I added enough debugging information in the logs, a weirder issue comes up that leads to me going "ugh I wish I added X piece of information to the log output".

I work on this app kinda sporadically, but I'll keep it in mind when I get a chance to work on this again. ty for the detailed bug report.

Damglador, 2024-09-26:
It definitely does something after pressing Yes. While I was connected to my phone with scrcpy and tested it, it restarted connection and took away the ability to use my laptop keyboard to with the phone. Probably totally useless information, but maybe not

Damglador, 2024-09-27:
I'm here to test at any time

bill88t, 2024-10-08:
Got it happening on magisk too..

  -  Device: Moto G32
  -  OS: Android 14
  -  Soft Keyboard: Unexpected
  -  Are You Rooted? Yup
  -  Root Method: Magisk

[debug_log_me.arianb.usb_hid_client_1728379608.txt](https://github.com/user-attachments/files/17291105/debug_log_me.arianb.usb_hid_client_1728379608.txt)
[image]

Zekiu, 2024-10-20:
> Got it happening on magisk too..
>
>   -  Device: Moto G32
>   -  OS: Android 14
>   -  Soft Keyboard: Unexpected
>   -  Are You Rooted? Yup
>   -  Root Method: Magisk [...]

Seems like it's the same issue that I had on my Moto g(100), where /dev/hidg0 is on /dev/hidg1 and /dev/hidg1 is on /dev/hidg2, I think it would be good if the app allowed to either change the path to these character devices, or add an option to use hidg1 and hidg2 instead of hidg0 and hidg1

Arian04, 2024-10-20:
> I think it would be good if the app allowed to either change the path to these character devices, or add an option to use hidg1 and hidg2 instead of hidg0 and hidg1

I've been meaning to add this feature for a long time but tbh I've just been busy, I got a midterm on Tuesday and lots of other stuff to work on, so I'll try to keep this in mind, but it might be a while until I can work on it

alr86, 2024-11-22:
Having the same issue.

Device: Redmi note 9
OS: Android 14 (CrDroid v10.9)
Soft Keyboard: FlorisBoard
Are You Rooted?: Yes
Root Method: Magisk

Log file: [debug_log_me.arianb.usb_hid_client_1732259042.txt](https://github.com/user-attachments/files/17866718/debug_log_me.arianb.usb_hid_client_1732259042.txt)

[image]

Driv28, 2024-12-24:
Same issues here. Xiaomi 14

[image]

[debug_log_me.arianb.usb_hid_client_1735082094.txt](https://github.com/user-attachments/files/18242129/debug_log_me.arianb.usb_hid_client_1735082094.txt)

Arian04, 2024-12-28:
Okay so I looked through all the logs and compared the issues. There seem to be two groups of issues:
1. /dev/hidg0 is missing, /dev/hidg1 looks fine
2. Both character devices are missing

To try and avoid asking everyone multiple times, I'm going to attach a shell script to retrieve just about every piece of information I think I'll want to know. Hopefully this means that once people start responding, I'll be able to figure out what's wrong without too many delays due to back-and-forth communication.

Please make sure to run the script in a root shell and respond with the output attached. Most terminals have some option to save the entire buffer into a text file, which could help you with this. The beginning of the script has usage instructions, the TL;DR of those instructions are "Just get the script onto the device, put it in /sdcard/ if you want, and run `source path/to/the/script`. Don't worry about making it executable."

GitHub won't let me attach the script as a file, so I'm putting it here as a code block:
```sh
#!/system/bin/sh

# Usage Notes:
#
# It's a bit annoying to find a location to put an executable script on android
# due to restrictions on what directories are writable and which ones are able
# to hold executable files.
#
# Because of this, I wrote this script with `source`-ing it in mind.
#
# In other words, just get the script onto your device and run:
#
# `source path/to/the/script/get_more_info.sh`
#
# Of course, replace that path with wherever you put the script. I just put it in /sdcard/

# Wrap the whole thing in () to run it in a subshell, so it's safe to `source` without
# inadvertedly altering the parent shell.
(
	set -x

	if [ ! $(id -u) -eq 0 ]; then
		echo "Please run the script as root :)"
		exit
	fi

	# Print every gadget recursively
	ls -lZR /config/usb_gadget/

	echo "------------------------------"

	ls -lZ /dev/hid*

	echo "------------------------------"

	ls -lZR /sys/class/udc
	getprop sys.usb.controller

	echo "------------------------------"
)
```

Driv28, 2024-12-28:
> Okay so I looked through all the logs and compared the issues. There seem to be two groups of issues:
> 
> 1. /dev/hidg0 is missing, /dev/hidg1 looks fine
> 2. Both character devices are missing
> 
> To try and avoid asking everyone multiple times, I'm going to attach a shell script to retrieve just about every piece of information I think I'll want to know. Hopefully this means that once people start responding, I'll be able to figure out what's wrong without too many delays due to back-and-forth communication.
> 
> Please make sure to run the script in a root shell and respond with the output attached.
> 
> **EDIT: Please also make sure to run this script after opening the app and attempting to create the character devices via the prompt and trying to send keystrokes or touchpad inputs.**
> 
> Most terminals have some option to save the entire buffer into a text file, which could help you with this. The beginning of the script has usage instructions, the TL;DR of those instructions are "Just get the script onto the device, put it in /sdcard/ if you want, and run `source path/to/the/script`. Don't worry about making it executable."
> 
> GitHub won't let me attach the script as a file, so I'm putting it here as a code block:
> 
> ```shell
> #!/system/bin/sh
> 
> # Usage Notes:
> #
> # It's a bit annoying to find a location to put an executable script on android
> # due to restrictions on what directories are writable and which ones are able
> # to hold executable files.
> #
> # Because of this, I wrote this script with `source`-ing it in mind.
> #
> # In other words, just get the script onto your device and run:
> #
> # `source path/to/the/script/get_more_info.sh`
> #
> # Of course, replace that path with wherever you put the script. I just put it in /sdcard/
> 
> # Wrap the whole thing in () to run it in a subshell, so it's safe to `source` without
> # inadvertedly altering the parent shell.
> (
> 	set -x
> 
> 	if [ ! $(id -u) -eq 0 ]; then
> 		echo "Please run the script as root :)"
> 		exit
> 	fi
> 
> 	# Print every gadget recursively
> 	ls -lZR /config/usb_gadget/
> 
> 	echo "------------------------------"
> 
> 	ls -lZ /dev/hid*
> 
> 	echo "------------------------------"
> 
> 	ls -lZR /sys/class/udc
> 	getprop sys.usb.controller
> 
> 	echo "------------------------------"
> )
> ```

Thank you.
I always get Following error when I try to run the script from /sdcard:

/system/bin/sh: /sdcard/USBHID.sh: can't execute: Permission denied

I have a rooted Xiaomi14and and used termux

Arian04, 2024-12-28:
I said to source it and not execute it to avoid that

alr86, 2024-12-28:
> Okay so I looked through all the logs and compared the issues. There seem to be two groups of issues:
> 1. /dev/hidg0 is missing, /dev/hidg1 looks fine
> 2. Both character devices are missing
> 
> To try and avoid asking everyone multiple times, I'm going to attach a shell script to retrieve just about every piece of information I think I'll want to know. Hopefully this means that once people start responding, I'll be able to figure out what's wrong without too many delays due to back-and-forth communication.
> 
> Please make sure to run the script in a root shell and respond with the output attached. 
> 
> **EDIT: Please also make sure to run this script after opening the app and attempting to create the character devices via the prompt and trying to send keystrokes or touchpad inputs.**
> 
> Most terminals have some option to save the entire buffer into a text file, which could help you with this. The beginning of the script has usage instructions, the TL;DR of those instructions are "Just get the script onto the device, put it in /sdcard/ if you want, and run `source path/to/the/script`. Don't worry about making it executable."
> 
> GitHub won't let me attach the script as a file, so I'm putting it here as a code block:
> ```sh
> #!/system/bin/sh
> 
> # Usage Notes:
> #
> # It's a bit annoying to find a location to put an executable script on android
> # due to restrictions on what directories are writable and which ones are able
> # to hold executable files.
> #
> # Because of this, I wrote this script with `source`-ing it in mind.
> #
> # In other words, just get the script onto your device and run:
> #
> # `source path/to/the/script/get_more_info.sh`
> #
> # Of course, replace that path with wherever you put the script. I just put it in /sdcard/
> 
> # Wrap the whole thing in () to run it in a subshell, so it's safe to `source` without
> # inadvertedly altering the parent shell.
> (
> 	set -x
> 
> 	if [ ! $(id -u) -eq 0 ]; then
> 		echo "Please run the script as root :)"
> 		exit
> 	fi
> 
> 	# Print every gadget recursively
> 	ls -lZR /config/usb_gadget/
> 
> 	echo "------------------------------"
> 
> 	ls -lZ /dev/hid*
> 
> 	echo "------------------------------"
> 
> 	ls -lZR /sys/class/udc
> 	getprop sys.usb.controller
> 
> 	echo "------------------------------"
> )
> ```
> 

```sh
# source /storage/emulated/0/sh.sh
[1735403486.804311] id -u
[1735403486.813246] '[' ! 0 -eq 0 ']'
[1735403486.813302] ls -lZR /config/usb_gadget/
/config/usb_gadget:
total 0
drwxrwx--- 6 root root u:object_r:configfs:s0  0 2024-12-28 19:57 g1

/config/usb_gadget/g1:
total 0
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-12-27 09:41 UDC
-rw-r--r--  1 root root u:object_r:configfs:s0  4096 2024-12-28 20:01 bDeviceClass
[... 164 lines cut ...]

/config/usb_gadget/g1/strings:
total 0
drwxrwx--- 2 root root u:object_r:configfs:s0  0 2024-12-28 19:57 0x409

/config/usb_gadget/g1/strings/0x409:
total 0
-rw-r--r-- 1 root root u:object_r:configfs:s0  4096 2024-12-27 09:41 manufacturer
-rw-r--r-- 1 root root u:object_r:configfs:s0  4096 2024-12-27 09:41 product
-rw-r--r-- 1 root root u:object_r:configfs:s0  4096 2024-12-27 09:41 serialnumber
[1735403486.825004] echo ------------------------------
------------------------------
[1735403486.825208] ls -lZ '/dev/hid*'
ls: /dev/hid*: No such file or directory
[1735403486.834424] echo ------------------------------
------------------------------
[1735403486.834451] ls -lZR /sys/class/udc
/sys/class/udc:
total 0
lrwxrwxrwx 1 root root u:object_r:sysfs:s0  0 2024-12-28 20:01 musb-hdrc -> ../../devices/platform/mt_usb/musb-hdrc/udc/musb-hdrc
[1735403486.843427] getprop sys.usb.controller
musb-hdrc
[1735403486.851091] echo ------------------------------
------------------------------
```

Arian04, 2024-12-28:
@alr86 thanks, and just to verify, did you run this right after opening the app and attempting to create the character device and send keystrokes (even if it expectedly fails)?

alr86, 2024-12-29:
> @alr86 thanks, and just to verify, did you run this right after opening the app and attempting to create the character device and send keystrokes (even if it expectedly fails)?

Yes👍

Arian04, 2025-01-03:
Looking over this again after I did some experimenting and wrote some testing code. Just wanted to let you guys know I edited your comments a bit to hide the code blocks in a little submenu to avoid cluttering the issue page as much. @Driv28 your output seemed slightly broken, so I also tried to clean it up a bit. Your version is saved in the edit history if you want to compare them or anything.

Arian04, 2025-01-06:
note #1: *meant to send this message a few days ago but got busy halfway through and forgot to finish it up*

note #2: just realized I accidentally messed up the formatting of Driv28's comment earlier, but just fixed it.

### Response to output received so far

@Driv28 : your output #3 looks very very close to functional actually. unless i missed something, it looks like only 1 symlink failed to be created, which is weird. 

@alr86 : app didn't seem to get far at all in creating the character device. the functions aren't even created in the usb gadget, which is extra strange because when you first submitted a comment on this issue, the app successfully detected hidg1 at least. Not sure what went wrong here.

### Code Progress Update

I have a version of the app with the (kinda fragile) shell script logic with a full root service that runs some Kotlin code with root permissions. I'm hoping this will achieve at least one of (but hopefully both) two things:
1. better error handling/error detection
2. more detailed debugging information when something goes wrong

I've been trying to bolt on more debugging info and working around the limitations of running a shell script in the app for gadget configuration for a while, now hopefully I can get entire detailed stacktraces and all that.

I'll drop a debug APK here sometime when I verify that the updated logic works as intended (as much as I can test that on one device).

alr86, 2025-01-06:
> @alr86 : app didn't seem to get far at all in creating the character device. the functions aren't even created in the usb gadget, which is extra strange because when you first submitted a comment on this issue, the app successfully detected hidg1 at least. Not sure what went wrong here.

As I remember, I used [[This app]](https://github.com/tejado/android-usb-gadget) in my first comment to manually enable `hid.gs0` but it didn't have any effect on the issue

wnccode, 2025-02-03:
hi everyone
i'm on Xiaomi Mi9T Pro/K20 Pro on soviet star kernel 4.14.352 with EvolutionX from 17/09/2024 (cant find version number ?!)
I had the same problem. no hidg0/hidg1 ok 
i just tried to rename hidg1 to hidg0 and hidg2 to hidg1 and everything fine ! got mouse and keyboard working through usb !!
 
i have to rename devices at each reboot and after creating devices with app, USB HID Client is always creating hidg1 and hidg2 not 0/1 on my device, like just starting index at 1 instead of 0 but both created devices (1/2) are fine and working. I don't know if it helps but i hope so

Arian04, 2025-02-04:
> hi everyone i'm on Xiaomi Mi9T Pro/K20 Pro on soviet star kernel 4.14.352 with EvolutionX from 17/09/2024 (cant find version number ?!) I had the same problem. no hidg0/hidg1 ok i just tried to [...]

That's great information thank you! I once again got super busy with life stuff, but when I do get time to work on this, my main issue is that my fixes for these bugs aren't thoroughly tested and are sometimes even slightly fragile. I might add some sort of experimental option to enable all my WIP code that may or may not function as expected. I'll try to get to it soon ish, but like I said, I've been busy.

Thanks again for this information!!

xfqwdsj, 2025-04-26:
I executed the `app/src/main/res/raw/create_char_devices.sh` and got these message. Maybe could help.

```
ln: cannot create symbolic link from '/config/usb_gadget/g1/functions/hid.keyboard' to '/config/usb_gadget/g1/configs/b.1//hid.keyboard': Invalid argument
ln: cannot create symbolic link from '/config/usb_gadget/g1/functions/hid.mouse' to '/config/usb_gadget/g1/configs/b.1//hid.mouse': Invalid argument
```

Arian04, 2025-06-06:
Hey everyone, I just released v3.0.0-beta, so I thought I should ping people. As explained in the release notes, I added support for some "device-specific quirks" and I also slightly modified the gadget configuration process.

Basically what I'm saying is:
1. please try out the v3.0.0-beta (following the update procedure at the top of the release notes)
2. see if there's still a problem
3. if so, then try to enable "Experimental mode", then enable the "Disable gadget functions during configuration" option that shows up underneath it, then reopen the app and try again.

At that point, if your issue is gone, please let me know, and if your issue is NOT gone, also please let me know.

These commits should at least fix the issue for users alryaz and szescxz, so I'm wondering how many other people's issues will be resolved by these changes.

Thank you!

Driv28, 2025-06-06:
> Hey everyone, I just released v3.0.0-beta, so I thought I should ping people. As explained in the release notes, I added support for some "device-specific quirks" and I also slightly modified the [...]

Wow Thank you! I can use the input now but I can't use the touchpad.

[debug_log_me.arianb.usb_hid_client_1749243409.txt](https://github.com/user-attachments/files/20634256/debug_log_me.arianb.usb_hid_client_1749243409.txt)

szescxz, 2025-06-07:
Rebooting with USB cable connected and without any of the experimental features:
- With USB debugging disabled:
Initial status of the gadget device is enabled, but after clicking "Yes“ on the "Character device(s) do not exist" prompt, "Troubleshooting" says "Failed to read permissions" on both character devices; the USB Gadget Tool app says the gadget device gets disabled, after enabling it there and allowing your app to fix the permissions everything works.
- With USB debugging enabled:
Everything works.

Arian04, 2025-06-08:
@szescxz thank you!! in response to the sidenote, permission fixing is *supposed* to be done when the character devices are first created, but a bug (that I just fixed a few minutes ago) made it so this was not happening. I made the error message very noticeable and repetitive because it shouldn't normally ever be triggered and indicates a problem.

woliver99, 2025-06-30:
The keyboard seems to be working fine but I don't have the touchpad working (not a big deal since i only needed the keyboard) but i noticed that hidg1 is missing but hidg2 gets created instead

[image]

[image]

Arian04, 2025-09-02:
@francois072 that text at the bottom that says your kernel doesn't have ConfigFS HID support means that your kernel is missing support for the feature that is required to make this app work. I can't fix that, you'll just have to flash a kernel with the feature enabled if you want to address it