Arian04/android-hid-client #54: Unexpected keyboard support

Zekiu, 2024-10-31:
**Is your feature request related to a problem? Please describe.**
Can't send function keys (F1-F12)

**Describe the solution you'd like**
Support for keyboards with function keys, like unexpected keyboard 

**Describe alternatives you've considered**

**Additional context**

Arian04, 2024-11-01:
cant reproduce this on the latest stable release of the app (I uninstalled my dev version and grabbed the stable version straight from the izzyondroid repo).

Please provide some more info on how to reproduce your issue. Did you create HID functions or modify them within another app (maybe USB gadget tool)? Because that would break things.

Cleanest way to test "fresh" is:
1. Reboot
2. Ensure the app has root privileges via whatever method you use (Magisk app, KernelSU app, etc.)
3. Open my app
4. Read the prompts and respond accordingly
5. Test it

After those steps, if it's still not working, let me know if any errors pop up on the screen and give me the info provided on the troubleshooting page.

Zekiu, 2024-11-01:
Ugh, it's seems like it just doesn't work in bios on my Dell optiplex 755 (although it does detected that a keyboard is connected, as I don't get a no keyboard error). The function keys work fine when booted to an OS, like Linux. Sorry for the troubles! I will close this issue now.

closed 2024-11-01 by Zekiu

Arian04, 2024-11-01:
it's alright! I might be able to reproduce this if I can find an old enough PC. I believe there are some nuances with adding support for HID devices before more advanced drivers can be loaded by an actual OS. I thought I tested that, but it's been a little while since I touched the keyboard report descriptor, so I can try to remember to verify that.