Arian04/android-hid-client #55: App Crashes immediately after agreeing to "Add HID funcitons to the default USB gadget?..."

anaxonda, 2024-11-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Install app
2. Allow superuser access
3. Tap yes to "Character devices(s) to not exist: Add HID funcitons to the default USB gadget?..." prompt

**Expected behavior**
doesn't crash

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device details (please complete the following information):**
 - Device: Samsung S10e
 - OS: Android 12 
 - Soft Keyboard: Floris Board Beta
 - Are You Rooted? Yes
 - Root Method: Magisk 26.3

**Additional context**
Add any other context about the problem here.

ProgrammerManSudjo, 2025-04-09:
I'm having the same issue as well, my specs are the following (it seems to be associated with Samsung devices):

**Device Details**

- Device: Samsung Note 9
- OS: Android 14
- ROM: Noble ROM 4.3
- Soft Keyboard: Unexpected Keyboard
- Are You Rooted? Yes
- Root Method: Magisk 28.1

ProgrammerManSudjo, 2025-06-17:
I apologize for the wait @Arian04, I've tried the following version `v.3.0.0-beta2` which didn't work, when launching the app at first it shows the "Character devices(s) to not exist: Add HID funcitons to the default USB gadget?..." prompt.
After declining or accepting the prompt it doesn't crash but it doesn't work, I've tried by disabling and enabling the experimental setting "Disable gadget functions during configuration" still it's the same result as previous.
And different installation procedures, updating and reinstalling as well.