Arian04/android-hid-client #56: Not working on Pixel 7a with Android 16 DP1

Eggmanplant, 2024-12-01:
**Describe the bug**
I can't use the app after upgrading my Pixel 7a to Android 16 DP1. Maybe because the kernel update to 6.1. The app can't read permissions of /dev/hidg0 and /dev/hidg1. The only file under /dev that contains "hid" in the name is /dev/uhid.

**To Reproduce**
1. Upgrade to Android 16 DP1
2. The app doesn't work now

**Expected behavior**
The app is working

**Device details**
 - Device: Google Pixel 7a
 - OS: Android 16 DP1, Pixel Stock
 - Soft Keyboard: Gboard
 - Is Rooted? Yes
 - Root Method: KernelSU

Arian04, 2024-12-27:
can you drop your debug log file here please?

Eggmanplant, 2024-12-28:
Here it is:
[debug_log_me.arianb.usb_hid_client_1735377963.txt](https://github.com/user-attachments/files/18265685/debug_log_me.arianb.usb_hid_client_1735377963.txt)

Vikramaditya015, 2025-04-06:
@Arian04 @Eggmanplant  I modified the script by adding some commands from usb gadget tool.
Instead of adding hid keyboard and mouse to default gadget, it creates a new usb gadget named "keyboard" and activates it . ( Same as how usb gadget tool works)

Some modifications is need in app to access the character device.
In my case it was hidg2 and hidg3

After these modifications, it works on pixel 7a with Android 16 beta 3.2

[create_char_devices.zip](https://github.com/user-attachments/files/19621973/create_char_devices.zip)

Arian04, 2025-04-06:
> Instead of adding hid keyboard and mouse to default gadget, it creates a new usb gadget named "keyboard" and activates it . ( Same as how usb gadget tool works)

this is very interesting, because I used to do this, but then using the character devices is mutually exclusive with stuff like USB tethering and such. never thought it would randomly break. this is great info and I guess I'll make this configurable.

A little while back, I refactored the gadget configuration code to be a root service and not just a bash script (bc of growing complexity like this) so I should be able to add support for this relatively easier than adding more logic onto the shell script (which will just grow unmanageable and/or fragile). that code is still on dev branch though because I have not polished it up or tested it thoroughly enough to feel comfortable making a stable release.

Eggmanplant, 2025-04-07:
I installed DP1, updated to DP2 and I can't use beta (or even stable) anymore. Updates do not show up, or they fail. I'm probably in some trans-Beta-DP state. I hate Google and their beta-DP-program-enroll-thing. Not using my Pixel anymore.

Vikramaditya015, 2025-04-07:
@Arian04 that's great, btw does the mouse(TouchPad) supports double tap and hold for draging

Arian04, 2025-04-07:
> btw does the mouse(TouchPad) supports double tap and hold for dragging floating windows? In release build I can't able to drag floating windows 

on the develop branch, I changed the touchpad to be a "real" touchpad rather than a mouse device, so if you grab that artifact from the "Actions" page then you should be able to try it. should work but the develop branch is for stuff I haven't thoroughly tested (which is really becoming a lot because it's been a while since I had a big chunk of time to dedicate to testing a bunch of situations on multiple physical devices, maybe I should just release it)

Eggmanplant, 2025-04-10:
Yeah. I did know.

Arian04, 2025-06-06:
Hey everyone, I just released v3.0.0-beta, so I thought I should ping people. As explained in the release notes, I added support for some "device-specific quirks" and I also slightly modified the gadget configuration process.

Basically what I'm saying is:
1. please try out the v3.0.0-beta (following the update procedure at the top of the release notes)
2. see if there's still a problem
3. if so, then try to enable "Experimental mode", then enable the "Disable gadget functions during configuration" option that shows up underneath it, then reopen the app and try again.

At that point, if your issue is gone, please let me know, and if your issue is NOT gone, also please let me know.

These commits have fixed a few people's issues already, so I'm wondering how many other people's issues will be resolved by these changes.

Thank you!

Arian04, 2025-06-13:
Thanks! I'm gonna mark this closed then