Arian04/android-hid-client #58: Losing ADB after creating devices

wnccode, 2025-02-11:
**Describe the bug**
Just after creating hid devices in app, my ADB access stop working. adn impossible to bring it back unless with a reboot

**To Reproduce**
Steps to reproduce the behavior:
1 start your phone
2 test adb connection (adb devices) => ok my device is listed
3 launch android-hid-client
4 devices are not present so app ask you to create them => click yes
5 devices created
6 retry ADB connection with command adb devices, device has gone

**Expected behavior**
to be able to use hid at the same time of other gadget (adb)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device details (please complete the following information):**
 - Device: Xiaomi K20Pro 
 - OS: Android 14 Evolution-X soviet-star kernel
 - Soft Keyboard: GBoard
 - Are You Rooted? YES
 - Root Method: KernelSU

**Additional context**
dmesg | grep adb returns after create devices and each time trying to start adbd
[ 1042.672750] type=1400 audit(1739311039.446:140756): avc:  denied  { open } for  comm="flags_health_ch" path="/dev/__properties__/u:object_r:adbd_prop:s0" dev="tmpfs" ino=1107 scontext=u:r:flags_health_check:s0 tcontext=u:object_r:adbd_prop:s0 tclass=file permissive=1
[ 1042.672806] type=1400 audit(1739311039.446:140757): avc:  denied  { getattr } for  comm="flags_health_ch" path="/dev/__properties__/u:object_r:adbd_prop:s0" dev="tmpfs" ino=1107 scontext=u:r:flags_health_check:s0 tcontext=u:object_r:adbd_prop:s0 tclass=file permissive=1
It's seems to be a selinux permission problem (not really sure of what I say, i admit that to first try troubleshooting by myself by reading your code so maybeeee i ask a bit to chatGPT and that said selinux can be involved there, just parroting a parrot here)

wnccode, 2025-02-12:
It seems to be specific to my config. I did a lot of debug stuff. but at one moment i just that maybe it is just my computer that can't see adb device cause of my hid as first descriptor. tried to chage update modify drivers to load all interfaces of usb composite device, but no luck. make it clean on my computer to avoid having a lot of not working drivers AND Voila! it just working...
hopefully i found my solution quite quickly.

It seem that my computer had an old winusb driver or something like that that i uninstalled at the same time as my zadig winusb driver or microsoft winusb etc after testing to update the driver

so my problem was my dirty computer that had the wrong driver

problem solved

closed 2025-02-12 by wnccode

Arian04, 2025-02-19:
I appreciate the update, thanks!