Arian04/android-hid-client #60: App not working in Android 10

Pecacheu, 2025-03-26:
**Describe the bug**
After clicking "Yes" to the "Character device(s) do not exist" popup, I get a message that reads `ERROR: Failed to create character device.`

In v2.2.1, the app continues to run, but unsurprisingly, the controls don't do anything. However in v2.3.0, the app just crashes outright after this point (Note: the app does not crash if I select "No" to the popup or deny root privileges.)

**To Reproduce**
Steps to reproduce the behavior:
1. Install the app
2. Open the app
3. Click "Yes" to the popup
4. Grant root privilege (if not already granted previously)
5. Error message
6. If v2.3.0, app crashes

**Device details (please complete the following information):**
 - Device: Fire HD
 - OS: LineageOS 17.1 (Android 10)
 - Soft Keyboard: Gboard
 - Are You Rooted? Yes
 - Root Method: TWRP + Magisk 26.1

Logfile:
[debug_log_me.arianb.usb_hid_client_1742938427.txt](https://github.com/user-attachments/files/19459317/debug_log_me.arianb.usb_hid_client_1742938427.txt)

ProgrammerManSudjo, 2025-04-09:
I have the same issue with my client. Same log file as well.

I've tried resseting my Magisk modules, deleting them. Factory resetting my phone as well and etc.

Arian04, 2025-06-06:
Hey everyone, I just released v3.0.0-beta, so I thought I should ping people. As explained in the release notes, I added support for some "device-specific quirks" and I also slightly modified the gadget configuration process.

Basically what I'm saying is:
1. please try out the v3.0.0-beta (following the update procedure at the top of the release notes)
2. see if there's still a problem
3. if so, then try to enable "Experimental mode", then enable the "Disable gadget functions during configuration" option that shows up underneath it, then reopen the app and try again.

At that point, if your issue is gone, please let me know, and if your issue is NOT gone, also please let me know.

These commits have fixed a few people's issues already, so I'm wondering how many other people's issues will be resolved by these changes.

Thank you!

Arian04, 2026-08-16:
I just released v3.1.0, if you have time to test that out, then feel free to let me know how it goes. If not, I will probably close this in a while due to a lack of response.

ProgrammerManSudjo, 2026-08-16:
Sadly still with v3.1.0, I'm experiencing the same error. It's the same error of "Character devices (s) do not exist".

Tried also by enabling Experimental Mode and "Disable gadget functions during configuration".

Arian04, 2026-08-18:
> Sadly still with v3.1.0, I'm experiencing the same error. It's the same error of "Character devices (s) do not exist".
> 
> Tried also by enabling Experimental Mode and "Disable gadget functions during configuration".

Hmm, weird. Could you send over the new log file?

Also, just to confirm, make sure that the "Disable gadget functions during configuration" setting is enabled, then reboot the phone, then try to see if creating the character devices works. This is to rule out the possibility of the failed attempt to create them possibly putting the usb gadget into a broken state.

Arian04, 2026-09-18:
Closing due to lack of response for 1 month. Feel free to reopen (or honestly even just open a new issue, as it becomes a bit messy with an old issue that was created by a different person with different logs) if you have time to update this. No worries if you don't

closed 2026-09-18 by Arian04