Arian04/android-hid-client #61: Not working on SM-T220 connected to Raspberry Pi

SuperCowProducts, 2025-04-13:
**Describe the bug**
Neither the touchpad nor the keyboard is working

**To Reproduce**

1. Connect Samsung T7 SSD to Raspberry Pi 5 in USB 3.0 slot (this is where my Raspberry's OS is located)
2. Use official PSU to power Raspberry Pi 5
3. Connect to VNC on some other device (this is to check if keystrokes are being registered)
4. Open USB HID Client app on SM-T220
5. Plug in USB cable
6. Click on "yes" (to accept device creation)
7. Attempt any of the following (manual/real-time keyboard/touchpad input)

**Expected behavior**
I was able to see my password being typed at the login screen on Raspberry Pi OS (obviously it works with a real USB keyboard)

**Screenshots**
N/A (please let me know if you need something specific, eg. System Logs etc.)

**Device details (please complete the following information):**
 - Device: [e.g. Google Pixel 5]
 - OS: Android 14, crDroid 10.11 (img src: https://github.com/naz664/crDroid_gsi/releases/download/v2024.12.19/crDroid-10.11-arm64_bgN-Unofficial.img.xz)
 - Soft Keyboard: Hacker's Keyboard, Gboard
 - Are You Rooted? Yes
 - Root Method: Magisk

**Additional context**
(I believe this is normal behavior that I encountered during testing)
- SM-T220 automatically reboots if I plug out the USB cable whilst using the app
- If developer options are enabled USB debugging will turn on when connecting SM-T220 to Raspberry Pi 5

<details open>
<summary>USB HID Logs</summary>
Do we have root permissions?: true
Root method: MAGISK

------------------------------
character device info for: /dev/hidg0
does it exist?: true
is it visible without root?: true
permissions: 
stdout (with extra newlines): 
crw-------
1
u0_a496
u0_a496
u:object_r:device:s0:c240,c257,c512,c768

485,

0
2025-04-13
08:27
/dev/hidg0

character device info for: /dev/hidg1
does it exist?: true
is it visible without root?: true
permissions: 
stdout (with extra newlines): 
crw-------
1
u0_a496
u0_a496
u:object_r:device:s0:c240,c257,c512,c768

485,

1
2025-04-13
08:27
/dev/hidg1

------------------------------
version: 4.19.191-28086179-abT220XXS7DXF9
has ConfigFS support?: true
has ConfigFS HID function support?: true
-
relevant snippet of kernel config: 
CONFIG_USB_CONFIGFS=y
CONFIG_USB_CONFIGFS_UEVENT=y
CONFIG_USB_CONFIGFS_SERIAL=y
CONFIG_USB_CONFIGFS_ACM=y
# CONFIG_USB_CONFIGFS_OBEX is not set
# CONFIG_USB_CONFIGFS_NCM is not set
# CONFIG_USB_CONFIGFS_ECM is not set
# CONFIG_USB_CONFIGFS_ECM_SUBSET is not set
CONFIG_USB_CONFIGFS_RNDIS=y
# CONFIG_USB_CONFIGFS_EEM is not set
CONFIG_USB_CONFIGFS_MASS_STORAGE=y
# CONFIG_USB_CONFIGFS_BICR is not set
CONFIG_USB_CONFIGFS_MTK_FASTMETA=y
# CONFIG_USB_CONFIGFS_F_LB_SS is not set
CONFIG_USB_CONFIGFS_F_FS=y
CONFIG_USB_CONFIGFS_F_DM=y
# CONFIG_USB_CONFIGFS_F_ADB is not set
CONFIG_USB_CONFIGFS_F_MTP=y
CONFIG_USB_CONFIGFS_F_PTP=y
CONFIG_USB_CONFIGFS_F_CONN_GADGET=y
CONFIG_USB_CONFIGFS_F_SS_MON_GADGET=y
CONFIG_USB_CONFIGFS_F_ACC=y
CONFIG_USB_CONFIGFS_F_AUDIO_SRC=y
# CONFIG_USB_CONFIGFS_F_UAC1 is not set
# CONFIG_USB_CONFIGFS_F_UAC1_LEGACY is not set
# CONFIG_USB_CONFIGFS_F_UAC2 is not set
CONFIG_USB_CONFIGFS_F_MIDI=y
CONFIG_USB_CONFIGFS_F_HID=y
CONFIG_USB_CONFIGFS_F_UVC=y
# CONFIG_USB_CONFIGFS_F_PRINTER is not set
# CONFIG_IIO_CONFIGFS is not set
CONFIG_CONFIGFS_FS=y

------------------------------
Logs: 
unknown_tag		INFO		Detected root method as: MAGISK
unknown_tag		INFO		Detected root method as: MAGISK
unknown_tag		INFO		Detected root method as: MAGISK
unknown_tag		INFO		Detected root method as: MAGISK
unknown_tag		INFO		Detected root method as: MAGISK
unknown_tag		INFO		Detected root method as: MAGISK
</details>

SuperCowProducts, 2025-06-15:
Unfortunately not! Congrats for the frequent updates though, I'm sure I'm very close to getting it work.

Here's my latest log:

[debug_log_me.arianb.usb_hid_client_1750018824.txt](https://github.com/user-attachments/files/20747381/debug_log_me.arianb.usb_hid_client_1750018824.txt)

Arian04, 2026-09-18:
Hi! Sorry for the incredibly late reply. I've been very busy and haven't really had much of a chance to work on the app for a while until now.

I actually have an unreleased version right now that adds a ton of debugging info (v3.2.0). I have it uploaded in this comment under another issue https://github.com/Arian04/android-hid-client/issues/83#issuecomment-5486846188

if you want to, please follow those instructions and provide a new debug log here when you've tried it. If you don't want to / don't have time, no worries.