Arian04/android-hid-client #63: App does not use the correct HID character devices for certain devices which create them at different paths than expected

Arian04, 2025-05-05:
Xiaomi Mi9T Pro/K20 Pro on soviet star kernel 4.14.352 with EvolutionX from 17/09/2024 (cant find version number ?!)

I had the same problem as #50 . (no hidg0/hidg1) 

i just tried to rename hidg1 to hidg0 and hidg2 to hidg1 and everything fine ! got mouse and keyboard working through usb !!

```sh 
mv /dev/hidg1 /dev/hidg0
mv /dev/hidg2 /dev/hidg1
# => fix permissions with USB HID Client app and everything working like a charm
```

> i have to rename devices at each reboot and after creating devices with app, USB HID Client is always creating hidg1 and hidg2 not 0/1 on my device, like just starting index at 1 instead of 0 but both created devices (1/2) are fine and working. I don't know if it helps but i hope so 

 _Originally posted by @wnccode in [#50](https://github.com/Arian04/android-hid-client/issues/50#issuecomment-2632135570)_

wnccode, 2025-05-05:
Hi Arian
I have a little update about my post.
The bug seems to be related to the ROM I used. I've updated to the last version which is now based on Lineage and devices have the right name now. The phone is still the same.
So not device related but ROM related.

Not very useful to give you that precision but may be It can help people who are looking for a phone that correctly name hid device

Arian04, 2025-05-07:
thank you, that's actually very useful info!

Arian04, 2025-06-06:
implemented in `develop` branch