Arian04/android-hid-client #66: Touchpad not working on windows

SzczurekYT, 2025-06-12:
**Describe the bug**
The keyboard works but the touchpad doesn't work on windows (but works on linux workstation). 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to a windows machine
2. Try using touchpad

**Expected behavior**
A clear and concise description of what you expected to happen.
Mouse moves

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device details (please complete the following information):**
 - Device: Nothing 3a
 - OS: Android 15
 - Soft Keyboard: Futo Keyboard
 - Are You Rooted? Yes
 - Root Method: Magisk

**Additional context**
The app version is the beta 3.0.0 since previous versions do not work for me due to #56
This is a laptop, maybe it is conflicting with the embedded, hardware touchpad?
(In such case it would be nice to have a pure mouse mode or smth like that)
I will not have a pc (without touchpad) until the end of this week to test, if this is correct assumption.

SzczurekYT, 2025-06-23:
> Same for me.
> 
> Dla Szczurka: To samo u mnie.
Thanks, I know English though xD

> EDIT: Does FUTO have a free version? If yes, I'll trash Gboard. Even if it's limited (I mean features, not time.)
Yeah, it's free, I have been happy with it as a gboard replacement

> Is "Linux Workstation" Fedora Workstation?
Almost, I use Nobara, but that's a fedora fork so.. almost identical stuff.

Arian04, 2025-08-27:
oops forgot to update that string after the testing happened. I'll do that when I can

SzczurekYT, 2025-08-28:
Tested a build from CI on windows, it's working now.

Arian04, 2025-08-29:
Cool! Thanks so much, I'll make a new issue specifically regarding Precision Touchpad Mode and the details regarding the missing feature reports.

closed 2025-08-29 by Arian04