Arian04/android-hid-client #67: Not working with bluetooth keyboard as input

KayJay7, 2025-06-13:
### Description

The application doesn't work with a bluetooth keyboard as "direct" input. Everything works with the regular on-screen keyboard as input, and the bluetooth one can write on the text box, but it doesn't do anything outside the textbox, even after pressing the "on-screen keyboard" button.

There is nothing on the logs, but I'll attach them anyway just in case there is something useful I didn't recognise.

This used to work in the past, but it doesn't now. I can't pinpoint when it stopped working as I upgraded both HID Client and Android

### To Reproduce
Steps to reproduce the behavior:
1. Connect a bluetooth keyboard
2. Connect to a PC via usb
3. Open HID Client
4. Try to type on the keyboard 
5. Nothing happens

### Expected behavior
The application should send the keyboard's keystrokes to the computer

### Device details:
 - Device: Google Pixel 6a
 - OS: Android 16
 - Soft Keyboard: Hacker's Keyboard, GBoard
 - Hardware Keyboard: Logitech K380 Bluetooth
 - Are You Rooted? Yes
 - Root Method: Magisk

### Additional context
Logs: [debug_log_me.arianb.usb_hid_client_1749826814.txt](https://github.com/user-attachments/files/20729519/debug_log_me.arianb.usb_hid_client_1749826814.txt)

Arian04, 2025-06-13:
interesting use case that I never considered, thanks for the bug report!

do you happen to remember which version this used to work on, and which one it stopped working on? and have you tested out the new beta (v3.0.0-beta)?

KayJay7, 2025-06-13:
I forgot to mention it, Yes, I am using 3.0.0-beta now, and I discovered now that it doesn't work. I think it used to work with either 2.3.0 or 2.2.1 with some older version of android.

KayJay7, 2025-06-13:
Thanks!

While we are at it, at least on some devices (I have tried both a pixel 6a with android 16, and a pixel 5 with android 14) if the USB configuration is set to "charge only" the app doesn't work at all, this might be worth mentioning on the README.

Arian04, 2025-06-13:
wow thank you so much for this! I actually immediately was able to reproduce this, but have never encountered it because I'm always developing this app with USB debugging enabled.

I think the gadget reset fails, but if you unplug and plug the device back in, it should probably reinit the gadget and fix itself (at least that's what I observed). I'll see how I can address this. I have a suspicion that an experimental "quirk" I implemented might just need to become default behavior.

so awesome that you brought this up to me though, because it would've been tough for me to catch since I always keep usb debugging and such enabled. thanks again!!

Arian04, 2025-06-15:
Hey! do you mind testing out the latest commit? I think it should fix your issue. I'm still working on making the `meta` key work though. I'm pretty certain Android is eating the key events because it wants to handle those as system shortcuts.

it should be available under [this](https://github.com/Arian04/android-hid-client/actions/runs/15666313779) CI run as the artifact named `app-release-unsigned`. The direct link is `here` if you don't want to have to look for it. It should download a zip archive, which you should be able to unzip to have the apk file.

Thank you in advance!

Arian04, 2025-06-18:
hmmm very interesting, thank you! I have a hard coded map to handle Android keycode->HID scan code mappings + handle which keys need to be sent to the device with a "shift" modifier (ex: you can type an asterisk on android without pressing the "shift key" first, but you'd generally need to on a physical keyboard). this map could be causing the problem.

I will try to reproduce this when I get a chance and figure out exactly what the cause of the problem is and what needs to be done to fix it (ie: refactor to not use that hard coded map, add logic for keyboard layouts if that's necessary, etc).

thanks for including so many details!!

Arian04, 2025-09-01:
Just an FYI, the keyboard layout issue is tracked in #39

Arian04, 2025-09-03:
So I know I just merged your issue into the other one, but I think I might've found a way to handle this that I think only works with physical keyboards. Was wondering if you'd like to test it using this zipped APK I just built with the required code.

[android-hid-client-v3.0.2_custom-release-build.zip](https://github.com/user-attachments/files/22125408/android-hid-client-v3.0.2_custom-release-build.zip)

Please let me know if things work properly (or even just change) when compared to your earlier tests in your comment [here](https://github.com/Arian04/android-hid-client/issues/67#issuecomment-2979947474).

Thank you!

Arian04, 2025-12-07:
sorry forgot to update you, but I committed the change and you should be able to grab the build here: https://github.com/Arian04/android-hid-client/actions/runs/19412527735

I'm also just gonna reopen this issue. I should've done so earlier, but I thought it would be a quick fix, but I spent longer on it than I expected. Thanks for your patience.

Edit: I also initially closed it because I moved handling of keyboard layouts to another issue. But the potential fix here only applies to physical keyboards, so I think keeping this as a distinct issue is fine.

Arian04, 2026-08-16:
I think I fixed this in v3.1.0 ([link here](https://github.com/Arian04/android-hid-client/releases/tag/v3.1.0)), let me know if you get a chance to test it. I tested this with my bluetooth keyboard way back then, but I just wasn't sure if I tested the layout handling stuff correctly. But unless i missed something during testing, it should hopefully work.

commit a73e1180 (2025-06-15): fix: regression that broke support for hardware keyboard handling in Direct Input

should fix #67