Arian04/android-hid-client #75: APK at v3.0.0 release

IzzySoft, 2025-08-29:
Are you sure you have the correct APK at that release? It claims to be built from f056958d1ae33a498d445639fa8a0e1d700823e6 – which is a 3.0.0.-beta2 from June. If that is what you intended, OK – that has now been distributed. If it was not, it might be time for a v3.0.1 😉 (Please do not replace what has been distributed already, but make a new release then, as the same release won't be processed twice, at several places not – e.g. at IzzyOnDroid, and most likely the same at F-Droid if your app is there).

Besides, building directly from that commit isn't reproducible either, but just because of [embedded build IDs](https://gitlab.com/IzzyOnDroid/repo/-/wikis/Reproducible-Builds/RB-Hints-for-Developers#no-funny-build-time-generated-ids) (please see that you can get rid of those, or RB cannot be achieved). So it's really that old Beta.

Arian04, 2025-08-29:
published that release, should be good now. Thanks again!

closed 2025-08-29 by Arian04

IzzySoft, 2025-08-29:
Well, let's take that chance to see if it makes the icon show up again… Yupp, was generated now. But your APK is no longer reproducible, as it now contains [buildIDs](https://gitlab.com/IzzyOnDroid/repo/-/wikis/Reproducible-Builds/RB-Hints-for-Developers#no-funny-build-time-generated-ids) in its native libraries:

[image]

Could you please apply the change suggested in above linked wiki page to your [`CMakeLists.txt`](https://github.com/Arian04/android-hid-client/blob/6bcf73441217993fa3f10486c2dca9e5cfbe1299/app/src/main/cpp/CMakeLists.txt)? That's the only thing keeping the APK from being reproducible.

Arian04, 2025-08-31:
yup, thanks for letting me know. I didn't have time to fix that one, so that's why I didn't fix it in the patch release, but I can do it in a future one. thanks again!

IzzySoft, 2025-08-31:
Thanks! Wasn't sure whether you got a notification for my previous comment, or might have missed that 😉